ottelo9/tasmota-sml-script #1: "Tasmota SML Script + Chart + PV.txt" reboot mit Exception:27 nach "kurzer Zeit" mit Tasmota 14.3.0

mi-hol, 2024-11-22:
Nach ca 2h habe ich immer wieder reboots des ESP32-C3 auf Tasmota 14.3.0

log:
```
00:00:00.250 CMD: Fall back to serial port, no SOF packet detected on USB port
00:00:00.250 HDW: ESP32-C3 v0.4 
00:00:00.261 UFS: FlashFS mounted with 288 kB free
00:00:00.269 CFG: Loaded from File, Count 83
00:00:00.276 QPC: Count 1
00:00:00.277 CFG: CR 401/699, Busy 0
00:00:00.278 SCR: ufilesystem found
00:00:00.305 SCR: nv=25, tv=3, vns=158, vmem=4099, smem=16384, gmem=588, tmem=21229 
00:00:00.307 TYA: Active=0
00:00:00.309 ROT: Mode 1
00:00:00.316 BRY: GC from 3898 to 3698 bytes, objects freed 4/17 (in 0 ms) - slots from 33/61 to 25/61
00:00:00.330 CFG: No '*.autoconf' file found
00:00:00.335 BRY: GC from 5826 to 4260 bytes, objects freed 8/35 (in 1 ms) - slots from 55/76 to 35/76
00:00:00.336 BRY: Berry initialized, RAM used 4260 bytes
00:00:00.341 BRY: No 'preinit.be'
00:00:00.343 SRC: Restart
00:00:00.346 Project tasmota - Tasmota Version 14.3.0(ottelo.jimdo.de)-3_1_0(2024-10-26T15:32:03)
00:00:00.350 SCR: performs "otaurl !!NEVER UPGRADE VIA WEB SERVER!!"
00:00:00.350 SRC: Rule
00:00:00.351 CMD: Grp 0, Cmd 'OTAURL', Idx 1, Len 32, Pld -99, Data '!!NEVER UPGRADE VIA WEB SERVER!!'
00:00:00.351 CFG: CR 401/699, Busy 0
00:00:00.352 RSL: RESULT = {"OtaUrl":"!!NEVER UPGRADE VIA WEB SERVER!!"}
00:00:00.364 SCR: performs "sensor53 r"
00:00:00.364 SRC: Rule
00:00:00.365 CMD: Grp 0, Cmd 'SENSOR', Idx 53, Len 1, Pld -99, Data 'r'
00:00:00.367 SML: uart used: 1
00:00:00.369 meters: 1 , decode lines: 3, memory used: 1816 bytes
00:00:00.370 RSL: RESULT = {"Time":"1970-01-01T00:00:00","SML":{"CMD":"restart"}}
00:00:00.390 ETH: No ETH MDC and ETH MDIO GPIO defined
00:00:00.392 SML: uart used: 1
00:00:00.393 meters: 1 , decode lines: 3, memory used: 1816 bytes
00:00:00.409 BRY: No 'autoexec.be'
00:00:00.421 NTP not ready
00:00:00.499 WIF: Checking connection...
00:00:00.499 WIF: Attempting connection...
00:00:01.002 WIF: Connecting to AP1 UPC2CB42C7 Channel 8 BSSId 90:5C:44:DA:9F:04 in mode HT40 as tasmota-1C972C-5932...
00:00:01.012 NTP not ready
00:00:02.150 WIF: IPv4 192.168.0.178, mask 255.255.255.0, gateway 192.168.0.1
00:00:02.736 NTP not ready
00:00:03.336 WIF: Checking connection...
00:00:03.337 WIF: Connected
00:00:03.423 WIF: IPv6 Local fe80::da3b:daff:fe1c:972c%st1
00:00:03.594 HTP: Web server active on tasmota-1C972C-5932 with IP address 192.168.0.178
00:00:03.736 NTP: Sync time...
00:00:03.771 WIF: DNS resolved '2.pool.ntp.org' (212.18.3.19) in 34 ms
00:00:03.814 RTC: UTC 2024-11-22T13:28:13Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
14:28:13.000 RTC: Synced by NTP
14:28:13.586 RSL: INFO1 = {"Info1":{"Module":"ESP32C3","Version":"14.3.0(ottelo.jimdo.de)","FallbackTopic":"cmnd/DVES_1C972C_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:28:13.599 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-1C972C-5932","IPAddress":"192.168.0.178","IP6Global":"","IP6Local":"fe80::da3b:daff:fe1c:972c%st1"}}
14:28:13.614 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":27,"Reason":"Unknown","EPC":"4203017c","EXCVADDR":"00004a96","CallChain":["420c0b42","420c0b42","4214c0e8","42152f10","420a033c","420a0300","4209b692","4214e090","40389d86","420308ae","42036c7c","42025dca","4214e090","40389d86","40389be2","40389d2c","420a8e86","420adf78","420ae07e","40389be2","40389d2c","42011ab6","42011ada","4207a0a2","4207a14e","4200d20a","42016710","40389c54","420adf78","40387b48","40387b48","4215cade","420d98f8","42086cf0","4207fe6c"]},"BootCount":40}}
14:28:13.659 TFS: File 'autoexec.bat' not found
14:28:16.423 WIF: IPv6 Global 2a02:8070:8883:1140:da3b:daff:fe1c:972c
14:28:16.725 QPC: Reset
14:28:17.728 BRY: GC from 5981 to 4607 bytes, objects freed 16/46 (in 1 ms) - slots from 72/91 to 50/91
14:28:17.730 RSL: STATE = {"Time":"2024-11-22T14:28:17","Uptime":"0T00:00:08","UptimeSec":8,"Heap":161,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"Wifi":{"AP":1,"SSId":"UPC2CB42C7","BSSId":"90:5C:44:DA:9F:04","Channel":8,"Mode":"HT40","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:28:17.756 RSL: SENSOR = {"Time":"2024-11-22T14:28:17","EasyM":{"Power_curr":205,"Total_in":89.74,"Total_out":3.69}}
14:28:18.734 APP: Boot Count 40
14:28:19.221 CFG: Saved, Count 84, Bytes 4096
14:28:22.737 RSL: config = {"ip":"192.168.0.178","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-1C972C-5932","mac":"D83BDA1C972C","md":"ESP32C3","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"14.3.0","t":"tasmota_1C972C","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"bat":0,"dslp":0,"sho":[],"sht":[],"ver":1}
14:28:22.742 RSL: sensors = {"sn":{"Time":"2024-11-22T14:28:22","EasyM":{"Power_curr":208,"Total_in":89.74,"Total_out":3.69}},"ver":1}
14:28:32.292 WIF: Checking connection...
14:28:52.339 WIF: Checking connection...
```

ottelo9, 2024-11-22:
Laut dem Script Entwickler gemu2015:
Im aktuellen Core scheint es ein Problem zu geben, wenn die Webseite länger braucht um zu laden. 

Ich melde mich falls ich mehr weiß.

mi-hol, 2024-11-23:
script läuft bisher 6h stabil auf "Tasmota 14.1.0 (ottelo.jimdo.de) by Theo Arends"

ottelo9, 2024-11-23:
Bei mir läuft „SML Script + Chart + PV„ mit Tasmota 14.3.0 seit 11 Tagen auf einem ESP32-C3 mini. Oft rufe ich die Webseite aber nicht auf, da alles über Home Assistant läuft.

mi-hol, 2024-11-23:
Ich brauche leider für ein Projekt dringend die Daten und habe im Moment nur einen Lesekopf, es wird deshalb ein bischen dauern bevor ich den Wunsch erfüllen kann.

nmW die schnellst Methode einen reboot zu provozieren, war bei mir die Zeile
`$<script> setTimeout("location.reload(true);",1000); </script>`
zu aktivieren

mi-hol, 2024-11-23:
script läuft bisher 14h30m stabil auf "Tasmota 14.1.0 (ottelo.jimdo.de) by Theo Arends"

mi-hol, 2024-11-28:
script läuft bisher 04d 18h30m stabil auf "Tasmota 14.1.0 (ottelo.jimdo.de) by Theo Arends"

@ottelo9  @gemu2015 wegen der Aussage "m aktuellen Core scheint es ein Problem zu geben, wenn die Webseite länger braucht um zu laden."  
gibt es dazu vielleicht einen passenden issue in https://github.com/arendst/Tasmota/issues?

ottelo9, 2024-11-30:
Danke für die Info! Woher bekommt man (wir als Nicht-Entwickler) solche Infos? Sind das nur Interna oder gibt es irgendwo ne guideline (oder in den Kommentaren)? 

Eigentlich schade, so muss ich wohl das Script Abspecken.

gemu2015, 2024-11-30:
nein das ist einfach ein Erfahrungswert. habe bei einigen Geräten festgestellt dass er hängt wenn es zu lange dauert. bei core 2 hat das aber nichts gemacht.
bei core 3 kommt er aber nicht zurück.

Man muss aber sicher stellen ob das wirklich daran liegt, oder sich sonst noch was geändert hat. Ich habe eine ganze Reihe von neuen Problemen seit core 3 gehabt.

du solltest also zunächst mal alle Grafiken herausnehmen und dann nacheinander wieder hinein bis der Fehler wieder auftritt. 
Dann mir Bescheid geben welcher Graf den Unterschied macht.

mi-hol, 2024-11-30:
> habe bei einigen Geräten festgestellt dass er hängt wenn es zu lange dauert. bei core 2 hat das aber nichts gemacht.
> bei core 3 kommt er aber nicht zurück.

 Linux hat da eine ganz klare Aussage bei solcher Art von Verhaltensänderungen "[We don’t cause regressions](https://www.kernel.org/doc/html/latest/process/handling-regressions.html) " ich öffne gleich ein issue um rauszufinden, ob das so bei Tasmota auch gilt.

gemu2015, 2024-11-30:
wie gesagt ich würde aber trotzdem erst gerne wissen was genau den reboot erzeugt, also welcher Programmteil des scripts.

mi-hol, 2024-12-01:
> Linux hat da eine ganz klare Aussage bei solcher Art von Verhaltensänderungen "[We don’t cause regressions](https://www.kernel.org/doc/html/latest/process/handling-regressions.html) "

habe das erst mal [in Discord gefragt](https://discord.com/channels/479389167382691863/479389167382691865/1312903369374830645)

der konkrete Fall existisitert schon in https://github.com/arendst/Tasmota/issues/22405

ottelo9, 2024-12-15:
So, ich habe meine Tests abgeschlossen. Es liegt nicht direkt an den Diagrammen, sondern anscheinend einfach an der Menge an HTML Zeichen. Gebe ich nur ein paar Wörter aus, so rebootet der ESP nie. Aber bei mehreren Sätzen gibts reboots. Abgespeckt hatte ich nur die >W Sektion. An einem Stromzähler hängt der ESP nicht.

Getestet mit diesem Script:
[reboot_script.txt](https://github.com/user-attachments/files/18140456/reboot_script.txt)

Ich verwende das hier damit die Seite alle 5s refresht:
$<script> setTimeout("location.reload(true);",5000); </script>

mi-hol, 2024-12-15:
> Getestet mit diesem Script:
> [reboot_script.txt](https://github.com/user-attachments/files/18140456/reboot_script.txt)

Super, hättest Du vielleicht einen ESP32 auf dem Du das mit der neuesten Tasmota 14.04 prüfen könntest?

mi-hol, 2024-12-15:
das letzte release 54 https://github.com/arendst/Tasmota/commit/da7473e07c39056421382f1cc29fe56d5ea0c671 ändert interessanterweise den WebSend Driver

ottelo9, 2024-12-15:
Teste ich nachher mal. Danke für die Info.

mi-hol, 2024-12-16:
Danke @ottelo9 !
Planst Du ein issue bei Tasmota zu öffnen?

ottelo9, 2024-12-16:
@mi-hol 
Erstmal nicht....
Ich bin gerade noch im direkten Kontakt mit @gemu2015 und teste ein paar Sachen...

mi-hol, 2024-12-25:
@ottelo9 magst Du vielleicht einen Status der Test mit gemu2015 publizieren?

ottelo9, 2024-12-26:
Ich hatte mit ihm mehrere Images getestet und habe alle Diagramme zum Test entfernt. Auch mit nur etwas Text in der >W Sektion kann man nach einigen Minuten einen Reboot provozieren, liegt also nicht direkt an den Diagrammen. Mehr Inhalt heißt häufigere Reboots.

Gemu meint, dass es wohl an einem oder mehreren Treibern liegen könnten, die seit 14.1.0 verändert wurden und den Absturz jetzt verursachen. Er deaktiviert immer alle unnötigen Treiber und bei ihm läuft das Image ohne Reboot.

Ich lasse aber immer gerne alles aktiv, damit man damit immer noch mehr machen kann als nur den Zähler auszulesen. Ich werde noch ein Image erstellen mit deaktivierten Treibern. (siehe undef Liste).

[undefs.txt](https://github.com/user-attachments/files/18253194/undefs.txt)

ottelo9, 2024-12-29:
Kurze Rückmeldung.
Ich teste gerade das Reboot Testscript mit Tasmota 14.3.0 und 14.4.1 auf einem ESP32-WROOM-32 und ESP01s. Scheint bislang zu laufen ohne Reboot.
[reboot_script.txt](https://github.com/user-attachments/files/18269757/reboot_script.txt)

mi-hol, 2025-01-13:
bei mehreren ModBus Abfragen nacheinander kam es wieder zum reboot.

Hilft dieser console log Eintrag?

```
01:46:47.464 RSL: INFO1 = {"Info1":{"Module":"ESP32C6","Version":"14.3.0(ottelo.jimdo.de)","FallbackTopic":"cmnd/DVES_2CF8C8_fb/","GroupTopic":"cmnd/tasmotas/"}}
01:46:47.467 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-2CF8C8-7724","IPAddress":"192.168.0.250","IP6Global":"","IP6Local":"fe80::f2f5:bdff:fe2c:f8c8%st1"}}
01:46:47.469 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":27,"Reason":"Unknown","EPC":"420300c2","EXCVADDR":"00005bf2","CallChain":["420c0ade"]},"BootCount":31}}
01:46:50.630 QPC: Reset
01:46:51.635 RSL: STATE = {"Time":"2025-01-13T01:46:51","Uptime":"0T00:00:08","UptimeSec":8,"Heap":254,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"Wifi":{"AP":2,"SSId":"UPC2CB42C7","BSSId":"90:5C:44:DA:9F:04","Channel":8,"Mode":"HT40","RSSI":24,"Signal":-88,"LinkCount":1,"Downtime":"0T00:00:03"}}
```

mi-hol, 2025-01-14:
```
SP32C6
SML_C6
Program Version	14.3.0 (ottelo.jimdo.de)
Build Date & Time	2024-11-12T11:28:44
Core/SDK Version	3_1_0/5.3.1.240924
```

@ottelo9 Habe ich da den richtigen Build geflasht?

mi-hol, 2025-01-14:
Wenn ich per ModBus Daten abfrage, bekomme ich den folgenden Fehler, mit einem 14.1.0 Image funktionert es.
Könnte das an einer Option liegen, die beim Compile angegeben wurde?

```
2025-01-14 13:52:08,137 | E | pymodbus.logging | Connection to (192.168.0.250, 502) failed: [WinError 10061] No connection could be made because the target machine actively refused it
2025-01-14 13:52:08,137 | E | modpoll.modbus_task | Failed to connect to Modbus client
```

ottelo9, 2025-01-16:
Was genau willst du denn nutzen?

1) Modbus SML um einen Zähler via Modbus TCP abzufragen?
2) Oder das? https://tasmota.github.io/docs/Modbus-Bridge

Ich kompiliere "eigentlich" immer mit diesen Features (alle Default Features/Treiber aktiviert):
https://github.com/ottelo9/tasmota-sml-script/blob/main/0_Tasmota_Image_erstellen_ESP32.txt

Ich kompiliere sonst einmal neu und lade sie erneut hoch für c6

mi-hol, 2025-01-16:
> Was genau willst du denn nutzen?

Sorry, auf die gute Frage, kann ich so direkt leider keine Antwort geben, da mir noch das Detailwissen fehlt :(

Ich will den ESP32 C6 auf dem "2 Tasmota SML Script + Chart + PV.txt" läuft, vom PC via tool modpoll abzufragen.
Ich lade einfach das neue C6 Image von GDrive und melde mich mit Ergebnis zurück.

PS:
Super, dass Du auch die Anleitung fürs Kompilieren jetzt hier hochgeladen hast.

PPS:
Man kann sich bei GH bei neuen Releases und anderen Triggern  automatisch benachrichtigen lassen :) 
Würde GH vielleicht auch als Download Quelle für images deshalb besser als der GDrive funktionieren?

mi-hol, 2025-01-16:
Ergebnis:

C6 mit Tasmota 14.3.0 (Ottelo)  => Fehler
C3 mit Tasmota 14.1.0 (Ottelo)  => ok

Im als Basis für mein script verwendeten script [Fronius Gen24 Smart Meter Modbus TCP SunSpec Emulation mit ESP32 und Tasmota](https://pastebin.com/11zX4kG5) das Du auf https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen-tasmota/ verlinkt hast, sind keine Kommandos für ModBus (zB ModbusTCPStart) enthalten, trotzdem hat Zugriff per modpoll funktioniert.
Hat sich da vielleicht beim OS Versionswechsel von 14.1 auf 14.3 etwas verändert?

```
PS C:\Users\mholzer\Documents\0_BalkonPV_Technik\StromzählerStefan\tests>  modpoll --tcp 192.168.0.250 --config FroniusGen24.string_bug.csv -1

Modpoll v1.4.1 - A New Command-line Tool for Modbus and MQTT

2025-01-16 11:44:45,901 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2025-01-16 11:44:45,901 | I | modpoll.modbus_task | Loading config from: FroniusGen24.string_bug.csv
2025-01-16 11:44:45,912 | I | modpoll.modbus_task | Added 1 device(s)...
2025-01-16 11:44:45,912 | I | modpoll.main | Loaded 1 Modbus config(s).
2025-01-16 11:44:45,913 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.0s ===
2025-01-16 11:44:48,430 | E | pymodbus.logging | Connection to (192.168.0.250, 502) failed: [WinError 10061] No connection could be made because the target machine actively refused it
2025-01-16 11:44:48,430 | E | modpoll.modbus_task | Failed to connect to Modbus client
PS C:\Users\mholzer\Documents\0_BalkonPV_Technik\StromzählerStefan\tests>  modpoll --tcp 192.168.0.178 --config FroniusGen24.sunspec.csv -1

Modpoll v1.4.1 - A New Command-line Tool for Modbus and MQTT

2025-01-16 11:45:16,378 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2025-01-16 11:45:16,378 | I | modpoll.modbus_task | Loading config from: FroniusGen24.sunspec.csv
2025-01-16 11:45:16,396 | I | modpoll.modbus_task | Added 1 device(s)...
2025-01-16 11:45:16,396 | I | modpoll.main | Loaded 1 Modbus config(s).
2025-01-16 11:45:16,396 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.0s ===

Device: FroniusGen24Sim
+-----------+-------+------+
| Reference | Value | Unit |
+-----------+-------+------+
| Version   |  None |      |
| Address   | 17237 |      |
| ID        |     0 |      |
| L         | 17144 |      |
| sml1_W    |   0.0 | Watt |

```

mi-hol, 2025-01-16:
> ich kompiliere sonst einmal neu und lade sie erneut hoch für c6

Damit ich auf dem C6 beide OS Versionen testen kann, Wärst Du bitte so nett, die Tasmota 14.1 für C6 auch nochmals kompilieren und auf den GDrive laden?

ottelo9, 2025-01-16:
Ich habe mir [CherAlba's script](https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script/blob/main/sml-modbus-mqtt.tas) mal angeschaut um rauszufinden, welche Tasmota Features benötigt werden. Im Script wird wswa() wsa() verwendet, das sind TCP Server Script Commands und benötigen das Feature USE_SCRIPT_TCP_SERVER (siehe auch https://tasmota.github.io/docs/Scripting-Language/). Das habe ich nämlich nicht mit rein genommen bei 14.3.0, da optional und sonst niemand genutzt hat. Deshalb kann das Script auch nicht auf ModbusPoll Anfragen reagieren.

Ich kompiliere eben 14.3.0 mit USE_SCRIPT_TCP_SERVER und lade es mit extra Dateinamen hoch.

mi-hol, 2025-01-16:
> Im Script wird wswa() wsa() verwendet, das sind TCP Server Script Commands und benötigen das Feature USE_SCRIPT_TCP_SERVER

Danke, Du bist mein "Held des Tages", das hätte ich nicht so einfach gefunden

Was meinst Du zu [obigem Vorschlag in "PPS"](https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2595135518) GH Releases für images zu nutzen?

ottelo9, 2025-01-16:
Die Images sind hochgeladen.

GH als reines Datengrab verwenden? Ich müsste dann ein neues Repo aufmachen für die Tasmota Images ... kann man natürlich machen, bei Gelegenheit...

mi-hol, 2025-01-16:
> GH als reines Datengrab verwenden?

Nein, denn das für die images essentiell wichtige Config file **undef.txt** wäre damit [bei Änderungen wie oben](https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2562748038) für alle sichtbar und solche Fragen wie "was hat sich geändert", könnte jeder der lesen kann, damit selbst beantworten

mi-hol, 2025-01-17:
Also auf einem ESP32C6 mit Tasmota 14.3.0 (ottelo.jimdo.de)
kann ich nach etwas mehr als 5 Minuten einen Reboot provozieren
Restart Reason	Timer Group0 Watch dog reset digital core

Könnten wir vielleicht ein 14.4.1 Image mal probieren?

```
Uptime	0T00:00:23
Flash Write Count	98
Boot Count	37

Uptime	0T00:04:43
Flash Write Count	97
Boot Count	36

Uptime	0T00:00:56
Flash Write Count	97
Boot Count	36
Restart Reason	Timer Group0 Watch dog reset digital core

Program Version	14.3.0 (ottelo.jimdo.de)
Build Date & Time	2025-01-16T12:02:13
Core/SDK Version	3_1_0/5.3.1.240924
Uptime	0T00:03:32
Flash Write Count	96
Boot Count	35
Restart Reason	Timer Group0 Watch dog reset digital core

```

ottelo9, 2025-01-17:
14.4.1 hat das Problem immernoch, hatte ich bereits getestet...

Ich habe eine Diskussion aufgemacht (hat mir Jason2866 geraten).

mi-hol, 2025-01-17:
aha, ist es diese https://github.com/arendst/Tasmota/discussions/22833 ?

mi-hol, 2025-01-17:
aus dem log in der discussion sehe ich `BRY: GC from 5559 to 4607 bytes` als eine der letzten Einträge.

"BRY" steht doch für Berry Script, oder?
Falls dein script das auch nicht braucht, würde ich vorschlagen als Test, Berry aus dem build raus zu nehmen.
Was meinst Du dazu?

ottelo9, 2025-01-17:
Das wollte ich auch, ist aber etwas komplizierter als gedacht. Und die Entwickler/Supporter haben mir davon dringend abgeraten. Nur wenn alle Stränge reißen. Aber das werde ich trotzdem mal versuchen … wenn ich Zeit habe.

mi-hol, 2025-01-17:
https://github.com/arendst/Tasmota/discussions/22833#discussioncomment-11871782 hat noch einen guten Vorschlag.
Ist der vielleicht einfacher umzusetzen?

mi-hol, 2025-01-17:
Um meine Theorie der potentiellen Fehlerursache zu überprüfen, wäre es noch gut auf einem ESP32-S2 (**Single-core** Xtensa LX7 CPU, up to 240 MHz ) einen Gegencheck zu machen. Ich hab leider nur kein Board mit der CPU :(

mi-hol, 2025-01-18:
Ich habe vielleicht gestern abend einen Workaround gefunden

statt:
; create task 1 every 10ms on core1, priority 3, stack 8kb
ct(1 10 1 3 8000)

jetzt:
; create task 1 every 50ms on core1, priority 3, stack 8kb
ct(1 50 1 3 8000)

damit habe ich jetzt schon Uptime: 0T19:22:43

ottelo9, 2025-01-18:
Das hört sich recht gut an. Wofür ist der Task 1 zuständig? Alle 10ms ist ja brutal schnell.

mi-hol, 2025-01-18:
das stammt aus dem ModBus script von @CherAlban @gemu2015  und beantwortet Anfragen 
https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script/blob/6b2a064ba1bb156d70b51b1643ae6647c10430bd/sml-modbus-mqtt.tas#L60

ottelo9, 2025-01-18:
Ah, ok ich dachte du bist im Tasmota Sourcecode fündig geworden. Das wird mir leider nicht mit meinen Scripten weiterhelfen ....

mi-hol, 2025-01-18:
ein guter Tipp fürs debugging https://github.com/arendst/Tasmota/discussions/20430#discussioncomment-11874705

mi-hol, 2025-01-18:
> ein guter Tipp fürs debugging [arendst/Tasmota#20430 (reply in thread)](https://github.com/arendst/Tasmota/discussions/20430#discussioncomment-11874705)

führt leider zu einem neuen Fehler auf C6. siehe https://github.com/arendst/Tasmota/issues/22839

mi-hol, 2025-01-18:
Ich gewinne immer mehr den Eindruck, dass es für die RISC-V CPUs oder Single Core CPUs,  gravierende Fehler im low level code gibt

mi-hol, 2025-01-18:
> führt leider zu einem neuen Fehler auf C6. siehe [arendst/Tasmota#22839](https://github.com/arendst/Tasmota/issues/22839)

Wärst Du so nett `SetOption130 1` mal zu probieren?
Bei https://github.com/arendst/Tasmota/issues/22839#issuecomment-2599764537 würde eine Bestätigung sehr helfen

ottelo9, 2025-01-18:
Bei mir funktioniert es. Den Log lasse ich mir via Web Console und Serial ausgeben. Script läuft dabei.

mi-hol, 2025-01-18:
Danke.

Ich habe mir durch Cody.AI mein Tasmota script auf Berry konvertieren lassen und probiere jetzt als nächstes, ob das stabiler läuft

ottelo9, 2025-01-18:
Bietet Berry denn alles was gemus Script bietet? Und wie webpage kann man auch bedienen?

mi-hol, 2025-01-19:
Werde ich sagen können, nachdem das konvertierte script läuft. Soweit ist es noch nicht.
Ich wechsle jetzt erst mal den C6 gegen einen C3

gemu2015, 2025-01-20:
ich wollte noch mal betonen das mit meinem C3 build nichts abstürzt.
Habe das jetzt etliche Tage ohne Probleme laufen lassen
Es liegt definitiv nicht an scripting oder dem SML Treiber

mi-hol, 2025-01-20:
@gemu2015 @ottelo9 ihr habt Recht, das [aktuelle script](https://github.com/ottelo9/tasmota-sml-script/blob/main/2_Tasmota_SML_Script_Chart_PV.txt) läuft perfekt!
Viele Dank dafür!

Es liegt an meiner lokalen Ergänzung aus https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script/blob/6b2a064ba1bb156d70b51b1643ae6647c10430bd/sml-modbus-mqtt.tas#L60 und dem Wechsel auf C6. 
Mein Fehler war den gleichen Scriptnamen  weiter zu verwenden. 
Das Ziel war eine Kombination von beiden (SML_Script_Chart_PV + FroniusSmartMeterSim) , aber das scheint nicht zu funktionieren.

Asche auf mein Haupt. Es tut mir leid!
Hoffe ich kann das mal wieder gutmachen

ottelo9, 2025-01-20:
Ja, ich hatte alle Skripte, nach einem Tipp von gemu, aktualisiert, um das Rebootproblem deutlich zu entschärfen. Ich hatte "if wm>0 { ... wcs google chart ... }" in die #subs eingefügt, damit diese nicht 4x die Sek gerufen werden, sondern nur, wenn die Seite geöffnet/aktualisiert wird.

Trotzdem habe ich mit dem Script auf meinem C3 mit meinem Image Rebootprobleme (alle 5min), nur allein wenn ich mit dem Webbrowser auf der Seite des ESPs bin und die Charts angezeigt werden (kein Aktualisieren der Seite). Wenn ich so gut wie alle Treiber/Features aus meinem Build entferne (BERRY kann ich bislang nicht ohne Compileerrors entfernen), dann kommen die Reboots nur noch alle 25min.

Ich kompiliere mit GitPod. @gemu2015 mit Visual Studio. Ich versuche es nun auch mit Visual Studio (Portable). Und ich mache den Test heute Abend mit dem Image von @gemu2015 (Tasmota 14.4.1).

mi-hol, 2025-01-20:
> Image von [@gemu2015](https://github.com/gemu2015) (Tasmota 14.4.1).

@gemu2015 kann man dein(e) Image(s) irgendwo runterladen?

ottelo9, 2025-01-20:
Ok nicht 4x die Sekunde sondern alle 2.5s (kleiner Unterschied) :)

> aber müsste man das nicht auch für die 4h & 24h

Die gleiche Frage hatte ich auch. Hier die Antwort:

> in den web subroutinen brauchst du die Abfrage
> 
> in der >W section macht das das vorangestellt $ Zeichen.

Das Image von gemu lade ich dir nachher hier hoch.

mi-hol, 2025-01-20:
> Hier ist das Image.
> [tasmota32c3-4M.gemu.zip](https://github.com/user-attachments/files/18481429/tasmota32c3-4M.gemu.zip)

ist tcpserver darin aktiviert?

ottelo9, 2025-01-21:
> ist tcpserver darin aktiviert?

Weiß ich nicht.

gemu2015, 2025-01-21:
@mi-hol 
du hast nicht denselben Fehler wie @ottelo9 
für alle Cx Prozessoren oder welche mit nur einem Core ist dein script zu zeitkritisch.
insbesondere den Task alle 10 Millisekunden laufen zu lassen ist problematisch.

generell sind die Single core Prozessoren und erst recht die Cx Prozessoren recht langsam.
Ich würde immer den ESP32-S3 wählen wenn es um zeitkritische Anwendungen geht.
Der ist mit Abstand der leistungsfähigste Prozessor der Reihe.

nach wie vor kann ich den Fehler von @ottelo9 nicht reproduzieren mit meinem C3
meiner läuft mit dem Testscript den ganzen Tag ohne reboot.

@ottelo9 
wenn du einen crash mit meinem script hast, schick mir den crash dump.
dann kann ich die Quelle des Crashs ermitteln. (das kann ich nur mit von mir kompilierten Versionen)

@mi-hol 
hier die Version mit TCP_server.
[tasmota32c3-4M.bin.zip](https://github.com/user-attachments/files/18486088/tasmota32c3-4M.bin.zip)

ottelo9, 2025-01-21:
@gemu2015 
Ich habe dir über Creationx geschrieben. Ich habe aber mein SML PV Script dafür verwendet und dir auch das Log mit angehängt (oder brauchst du was anderes?).

Ich teste das ganze nochmal mit dem Testscript, das du mir via PM geschickt hattest.

Es kann vielleicht auch am ESP Lesekopf von Hichi liegen. Es ist ein ESP32 C3 Mini Modul drauf.

mi-hol, 2025-01-21:
> [@mi-hol](https://github.com/mi-hol)
> hier die Version mit TCP_server.

Danke.
Kleines Problem bei mir, C3 ist gerade nicht in meiner Bastelkiste zu finden, nur ein ESP32-WROOM 
ESP Chip Id | 6354548 (ESP32-D0WDQ6 v1.0)
-- | --
liegt für den vielleich auch ein passendes Image "rum"?

[30 more replies]

closed 2025-02-01 by ottelo9