ottelo9/tasmota-sml-script #3: Speicherung der build Konfiguration eines Image in GH

mi-hol, 2025-01-17:
> GH als reines Datengrab verwenden?
> 
> Nein, denn das für die images essentiell wichtige Config file **undef.txt** wäre damit [bei Änderungen wie oben](https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2562748038) für alle sichtbar und solche Fragen wie "was hat sich geändert", könnte jeder der lesen kann, damit selbst beantworten 

 _Originally posted by @mi-hol in [#1](https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2595638391)_

mi-hol, 2025-01-31:
Differenz von "Program Size", "Free Program Space" & "Free  Memory"
[14.3_tcpserver ](https://drive.google.com/file/d/1a9kh8pH2R8uYSQ25-a4od-g43NzPehNV/view?usp=drive_link) zu [14.4.1_ohne Domoticz](https://drive.google.com/file/d/1-LqzkczCfzftg1zB0W90MFLlV2Vfjnu3/view?usp=drive_link)

[image]

ottelo9, 2025-01-31:
Hier sind die Images für C3 und C6 zum Testen:
https://drive.google.com/open?id=1PsGNWf6VwCN-fdvi45bY8-8CoK_nEdf9&usp=drive_fs

Brauchst du eigentlich für Modbus Slave [Domoticz Support](https://tasmota.github.io/docs/Domoticz/)?
Siehe auch das Script https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script
[image]

So wie ich das verstehe, ich das so ähnlich wie Home Assistant? Aber die Einstellungen da verstehe ich nicht so ganz... 
Ich habe ein Image auch mal ohne USE_DOMOTICZ kompiliert. Kannst ja mal beide testen. Würde gerne so viel wie möglich aus dem Image raus lassen.

mi-hol, 2025-02-01:
> Hier sind die Images für C3 und C6 zum Testen: https://drive.google.com/open?id=1PsGNWf6VwCN-fdvi45bY8-8CoK_nEdf9&usp=drive_fs
> 
> Brauchst du eigentlich für Modbus Slave [Domoticz Support](https://tasmota.github.io/docs/Domoticz/)?

ist der TCPserver im C6 Image [14.4.1_ohne Domoticz](https://drive.google.com/file/d/1-LqzkczCfzftg1zB0W90MFLlV2Vfjnu3/view?usp=drive_link) aktiviert?
Ich bekomme Fehler:   [WinError 10061] No connection could be made because the target machine actively refused it

mi-hol, 2025-02-01:
> Update:
> Ich habe es gestern endlich hinbekommen, ein rebootfreies stark abgespecktes (ohne Matter, Berry und andere für den Lesekopf unwichtige Features) 14.4.1 Image für den ESP32-C3 (und die anderen Varianten) zu erstellen.

Ah, erst jetzt erst gesehen, das erklärt wohl meinen Fehler.

Ist das korrigierte C6 Image schon zum download verfügbar?

mi-hol, 2025-02-01:
> Ist das korrigierte C6 Image schon zum download verfügbar?

ich probiere jetzt mit https://drive.google.com/file/d/1-9zGERs_0GJaSh4XcH9QLHvKCsH16tBS/view?usp=drive_link

mi-hol, 2025-02-01:
Nach der Lösung von #5 könnte man mittels GH Actions direkt ein Image erzeugen und via GH releases publizieren.
Der Aufwand würde sich signifikant reduzieren.
Weiterer Vorteil, Jeder fork könnte dann eigene Varianten ganz einfach erzeugen

mi-hol, 2025-02-01:
Gute Beispiele für GH Actions könnten in https://github.com/FOME-Tech/fome-fw/tree/master/.github/workflows sein.

@ottelo9 Soll ich das ausprobieren?

ottelo9, 2025-02-01:
> ist der TCPserver im C6 Image aktiviert?

Ja.
Alle deaktivieren/aktivieren Features siehe: https://github.com/ottelo9/tasmota-sml-script/issues/5

mi-hol, 2025-02-01:
> Alle deaktivieren/aktivieren Features siehe: [#5](https://github.com/ottelo9/tasmota-sml-script/issues/5)

@ottelo9 ich _meine mal irgendwo gelesen zu haben_ , daß man die aktivierten Features via console auslesen kann.
Konnte dazu aber nichts in der Tasmota Dokumentation finden.
Habe ich mit das nur eingebildet?

ottelo9, 2025-02-01:
Status4

bzw. via https://github.com/tasmota/decode-status

Kann aber nicht alle Features/Treiber anzeigen.

**tasmota32c3_ottelo.bin:**
`STATUS4 = {"StatusMEM":"ProgramSize":1012,"Free":1867,"Heap":183,"StackLowMark":7,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":"0809","0740058A","00080001","00000001","00000000","00000008","00000080","00200000","04000000","00000000","00000000"],"Drivers":"1,2,!4,9,10,12,38,50,62,!68","Sensors":"1,53"}}`

ottelo9, 2025-02-01:
Also bei mir funktioniert es mit dem C3 (ohne Domoticz). Ich habe das [Script ](https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script/blob/main/sml-modbus-mqtt.tas) eingefügt. Via Modbus Poll kann ich mich verbinden und die Daten abfragen.

[image]

mi-hol, 2025-02-01:
> Also bei mir funktioniert es mit dem C3 (ohne Domoticz).

Dann probiere ich mal deinen Test, wenn es funktioniert, liegt der Fehler an meinem script.

> https://github.com/tasmota/decode-status

gibt die interessante Info "Reason":"Instruction access fault" für den C6.

Identifiziert die Build time unten, den korrigierten Build?

```
"StatusFWR":{"Version":"14.4.1(ottelo.jimdo.de)",
"**BuildDateTime":"2025-01-31T15:22:23**",
"Core":"3_1_1","SDK":"5.3.2.250106","CpuFrequency":160,"Hardware":"ESP32-C6FH4 v0.1","CR":"381/699"}
"StatusSTK":{"Exception":1,"Reason":"Instruction access fault","EPC":"30303030","EXCVADDR":"30303030","CallChain":["30303030"]}

```

mi-hol, 2025-02-01:
Weitere Erkenntnis:

mein Skript ist die Ursache

```
>BS
;Tasmota devices configured for native discovery via MQTT  (default)!
SetOption19 0
; start TCP server used for ModBus
res=wso(502)
print start TCP server - return code: %res%
```
gibt auf console:
`start TCP server - return code: -2.00`
und modpoll Abfrage bekommt:
` Connection to (192.168.0.81, 502) failed: timed out`

nehme ich `SetOption19 0` raus, bekomme ich:
`start TCP server - return code: 0.00`

und modpoll Abfrage bekommt Antwort.

Damit ist die offene Frage:
Sind SetOption nicht in allen Sektion zulässig?

mi-hol, 2025-02-02:
> mein Skript ist die Ursache

Der C6 hat eine "Power" und eine "Efficenx" CPU Core ( = "hybride" CPU), durch Änderung des für Modbus Task verwendeten CPU core zum schnellen 240Mhz PowerCore, scheint es gelöst zu sein.
Die von[ Linux bekannten Probleme](https://cse.google.com/cse?cx=partner-pub-0253814508491313:1305299758&ie=UTF-8&q=Hybrid+CPUs&sa=Search&ref=cse.google.com/) mit "hybriden CPUs" scheinen auch die ESPs zu treffen :(

Mein Fix ist in https://github.com/CherAlban/tasmota-sml-modbus-gen24-domoticz-ha-script/pull/4

mi-hol, 2025-02-02:
meine letzte Ergänzung verursacht: "SCR: init error -5" =  variable name length too long in total
; SCR: nv=42, tv=4, **vns=250,** vmem=4832, smem=16384, gmem=428, tmem=21894 
; vns = total size of name strings in bytes (may not exceed 255) 
 #**define SCRIPT_LARGE_VNBUFF** extents the size to 4095

könnte das vielleicht im Image geändert werden?

ottelo9, 2025-02-02:
Deine Variablen Namen sind 5 chars in Summe zu lang. Die Namen einfach etwas kürzer halten.

mi-hol, 2025-02-02:
> Die Namen einfach etwas kürzer halten.

Assembler, bei dem man kurze Namen verwenden musste, habe ich das letzte Mal ca 1988 verwendet :(
Das war auf einer IBM360 mit 256kB Main-Memory!
Dachte der Fortschritt bringt eine Verbesserung, Memory ist ja jetzt mehr als genug vorhanden.
Von den freien 284,4kB Memory würde ich 4kB gerne für lange und damit verständliche Variablennamen opfern, den damit wir der code selbstdokumentierend!

closed 2025-02-10 by ottelo9