ottelo9/tasmota-sml-script #4: falsche Berechnung von Tages-, Monats- & Jahreswerten für Bezug/Einspeisung bei Erstinstallation der Skripte

mi-hol, 2025-01-31:
**Use case:**

Skripts zeigt nur tatsächlich gemessene Werte ab Datum der Erstinstallation

**Zähler hat folgende Werte:**

Verbrauch (1.8.0):    4000 kWh
Einspeisung (2.8.0): 2000 kWh

Datum der Erstinstallation des Zählers:  10.10.2022
Datum der Erstinstallation des Skripts: 15.2.2025

**1.) Anzeige des Skripts  ab 60s nach der Installation:**

Tagesverbrauch:    0 kWh
Monatsverbrauch: 0 kWh
Jahresverbrauch: 4000 kWh

ottelo9, 2025-02-01:
; (c) Initialisieren
; Damit die Diagramme und Tabellen beim ersten Einrichten funktionieren folgendes in die Console auf eurem ESP eingeben (; am Anfang entfernen) - Achtung alle bereits gespeicherten Diagrammdaten werden gelöscht!
;backlog script >dval=EnFrGrid; script >dval2=EnToGrid; script >dcon={0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0}; script >dprod={0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0};
;backlog script >mval=EnFrGrid; script >mval2=EnToGrid; script >mcon={0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0};
**;backlog script >yval=EnFrGrid; script >yval2=EnToGrid;     Zaehlerstand beim Jahreswechsel. Statt EnFrGrid / EnToGrid koennt ihr dort auch euren alten Zaehlerstand eingeben, falls ihr diesen am Ende des Jahres oder Neujahr notiert habt (kWh).**

Im Script ist noch ein kleiner Bug.
if (secs%3==0)
if (secs%30==0)
if (secs%60==0)

tauschen gegen

if (upsecs%3==0)
if (upsecs%30==0)
if (upsecs%60==0)

werde ich natürlich noch korrigieren.

mi-hol, 2025-02-01:
> ;backlog script >yval=EnFrGrid; script >yval2=EnToGrid; Zaehlerstand beim Jahreswechsel.

Danke, nach dem x-ten mal installieren, habe ich daran nicht mehr gedacht :(
Ich denke mir dazu mal was "einfacheres" aus, das den Anwender "an die Hand nimmt" (zB Aufteilung in 2 scripte (Init & Run).

closed 2025-02-01 by ottelo9

mi-hol, 2025-02-02:
> tauschen gegen
> 
> if (upsecs%3==0) if (upsecs%30==0) if (upsecs%60==0)
> 
> werde ich natürlich noch korrigieren.

ich dachte das wolltest Du noch machen, bevor Du den issue schliesst