ottelo9/tasmota-sml-script #6: Grund für Definition der "Tages, Monats, Jahres Verbrauch & Einspeisung" Variables als permanente Variablen?

mi-hol, 2025-02-15:
im script https://github.com/ottelo9/tasmota-sml-script/blob/main/2_Tasmota_SML_Script_Chart_PV.txt#L63 

werden permanente Variablen benutzt, die aber dann mit 0 initialisiert werden.

@gemu2015
Die Tasmota Scripting Dokumentation ist leider sehr vage in dem Punkt "permanente Variablen" über Vor- und Nachteile, auch bleibt offen wie man sie zurücksetzt. Könntest Du vielleicht dazu was hier schreiben?
PS: ich verbessere die offizielle Tasmota Scripting Dokumentation laufend durch PRs, es würde dann dort eingearbeitet.

Doc:

_p:vname
specifies permanent variables. The number of permanent variables is limited by Tasmota rules space (50 bytes) - numeric variables are 4 bytes; string variables are one byte longer than the length of string. p vars are stored sequentially in the order of defintion. therefore when specifing permanent variables, add newly defined ones always at the end of already defined p vars. otherwise variables are mixed up and string variables may even be destroyed._

Ich habe die Reihenfolge von original:
```
; Tages, Monats, Jahres Verbrauch / Einspeisung
p:mval=0
p:dval=0
p:mval2=0
p:dval2=0
p:yval=0
p:yval2=0
```
auf folgendes geändert und bekomme jetzt seltsame Werte, die auch nicht ändern kann.
Mache ich da vielleicht was falsch?

```
; Tages, Monats, Jahres Verbrauch / FromGrid
; Note: p: = permanent storage! ?? why - reset to 0 at boot!
p:dvalF=0
p:mvalF=0
p:yvalF=0
; Tages, Monats, Jahres Einspeisung / ToGrid
p:dvalT=0
p:mvalT=0
p:yvalT=0

```

Console:
```

11:34:19.380 CMD: backlog script >dvalF=0; script >dvalT=0;
11:34:19.389 RSL: RESULT = {"Script":">dvalF=0"}
11:34:19.628 RSL: RESULT = {"Script":">dvalT=0"}
11:34:30.910 CMD: script?dvalT
11:34:30.912 RSL: RESULT = {"script":{"dvalT":"��B"}}
11:35:37.675 CMD: script?dvalF
11:35:37.677 RSL: RESULT = {"script":{"dvalF":"��B "}}

```

gemu2015, 2025-02-15:
das setzen auf 0 sagt dem interpreter hier lediglich dass es eine numerische Variable ist.
mit p:var="" würde eine Stringvariable definiert.
da permanente Variablen eben permanent sind werden sie beim initialisieren auf den permanenten Wert aus dem Flash gesetzt.

mit script>val=x

kannst du den wert in der Konsole setzen (kein Leerzeichen dazwischen)

den backlog kannst du dir fast immer sparen.
der macht nur Sinn wenn eines der zwischen backlog stehenden Befehle zu lange Zeit braucht.

gemu2015, 2025-02-15:
merke gerade dass der space doch erlaubt ist.
der backlog wäre aber nicht wirklich nötig da die Befehle hier schnell abgearbeitet werden
Schaden tut backlog aber auch nicht
keine Ahnung warum dein Beispiel oben nicht funktioniert. habe es gerade noch mal probiert, geht so wie es soll

mi-hol, 2025-02-15:
> keine Ahnung warum dein Beispiel oben nicht funktioniert. habe es gerade noch mal probiert, geht so wie es soll

bei mir ging es erst nachdem ich die Variablen als **nicht permanent** definiert habe.
Könntest Du bitte obiges Scenario mit Änderung der Reihenfolge und Name der permanenten Variable mal nachstellen?
Denke es ist möglicherweise so ein Grenzfall

gemu2015, 2025-02-15:
hab deine Variablen übernommen und das genauso eingegeben wie du oben
geht problemlos.

poste mal das komplette script

mi-hol, 2025-02-15:
es ist mMn die Reihenfolge der Aktionen entscheidend um den Fehler zu triggern.
Das ganze script hat für eine Reproduktion zu viel "noise". Ich versuche gelegentlich mal eine Reproduktion.
In welchem Repository poste ich das dann am besten, damit Du es gleich siehst?

gemu2015, 2025-02-15:
es könnte sein das du den maximalen permanenten Speicher überschritten hast. Es gibt dann leider keine Warnung, sondern es passieren solche Fehler wie oben.
insbesondere wenn du permanente Arrays  definiert hast geht schon viel p Speicher verloren.

bau doch einmal ein ganz kleines script mit nur den Variablen von oben und teste damit.

mi-hol, 2025-02-15:
> bei mir ging es erst nachdem ich die Variablen als **nicht permanent** definiert habe.

> es könnte sein das du den maximalen permanenten Speicher überschritten hast. Es gibt dann leider keine Warnung, sondern es passieren solche Fehler wie oben.
> insbesondere wenn du permanente Arrays definiert hast geht schon viel p Speicher verloren.

Das wäre eine gute Erklärung,.
Lässt sich das event "überschreiten maximalen permanenten Speicher" als ein Abbruch behandeln?
Für Script Entwickler ist das heute Verhalten aus meiner Sicht fatal und die schlechteste Implementierung!

mi-hol, 2025-02-15:
> event "überschreiten maximalen permanenten Speicher" als ein Abbruch behandeln?

so was wie "init error -5" bei vn wäre gut

mi-hol, 2025-02-15:
> es gibt allerdings in den docs eine Erläuterung dazu.

Habe nur die (aus mMn unvollständige) Erklärung (siehe copy&paste ganz oben) gesehen. Wo ist die Erläuterung  denn?

> für alle ESPs mit Filesystem ist es ohnehin besser dort die Variablen permanent zu speichern.

> Die Tasmota Scripting Dokumentation ist leider sehr vage in dem Punkt "permanente Variablen" über Vor- und Nachteile, auch bleibt offen wie man sie zurücksetzt. Könntest Du vielleicht dazu was hier [...]

Dein Kommentar ist wahrscheinlich ein der Teil der (zumindests mir) unklaren  Vor- und Nachteile.
Wenn ich eine komplette Liste habe, kann ich die gerne in die Doku einfügen

gemu2015, 2025-02-15:
in der doku steht doch wie viel permanenter Speicher vorhanden ist da gibt es eine Tabelle dazu
dann steht dort dass numerische Variablen 4 Bytes verbrauchen und String Variablen so groß wir die definierte Stringsize + 1
werd dir mal ein Beispiel raussuchen.
in meinen Dropbox download sind auch mehrere Beispiele.  (Link in den Docs)

mi-hol, 2025-02-15:
> in der doku steht doch wie viel permanenter Speicher vorhanden

Tut mir leid, aber bei "runtime" bzw "script spezifischen" Limits, habe ich ganz andere Erwartungen, als so was auf Grund von Doku manuell auszurechnen! Assembler mit solchem Verhalten habe ich ab ca 1985 deshalb nicht mehr angerührt!

> in meinen Dropbox download sind auch mehrere Beispiele. (Link in den Docs)

1. keine Treffer wenn man nach "dropbox" sucht
2. aus Sicherheitsgründen, lade ich nichts von file hostern! Git wurde aus dem Grund Transparenz und Sicherheit erfunden!

[image]

Für die anderen Tasmota Entwickler scheint aus obigen Gründen, die scripting language deshalb wogl ein "rotes Tuch" zu sein.
Können wir bitte wieder ernsthaft diskutieren?

mi-hol, 2025-02-16:
> werde mir mal den code ansehen ob ich das nicht abfangen kann.

könnte man in dem Zusammenhang, die Größe des Speichers für permanente Variablen, per Compiler Option, vergrößern?
Mein aktuelles Skript braucht voraussichtlich 3072 - 4096 Bytes

gemu2015, 2025-02-16:
nein, du hast nur so viel wie in der Tabelle in der Doc angezeigt wird also meistens 1536 Bytes
wie gesagt eh besser mit dem Dateisystem, insbesondere für Arrays. Geht natürlich nicht bei ESPs ohne Dateisystem.
würde einzelne Variablen mit p: definieren und arrays im Filesystem speichern.

Beispiel für das speichern von 4 Arrays:

```
#load
print loading arrays
dp2
fr=fo("/wd_log.txt" r)
fra(axhum fr)
fra(tmpa fr)
fra(aco2 fr)
fra(atvc fr)
fc(fr)

#save
print saving arrays
dp2
fr=fo("/wd_log.txt" w)
if fr>=0
then
	fwa(axhum fr)
	fwa(tmpa fr)
	fwa(aco2 fr)
	fwa(atvc fr)
	fc(fr)
else
	print file error: %fr%
endif
```

noch mal zur Verdeutlichung von backlog
backlog separiert einzelne Konsolen Kommandos in einer Zeile,
bei script brauchst du das nicht, sondern kannst einzelne Befehle einfach mit Semikolon trennen.
wird in einem Befehl abgearbeitet.

script>dvalF=0;dvalT=0; etc

PS. 
habe die Überlaufprüfung für permanente Variablen sowie Anzeige der Anzahl eingebaut (pmem)

00:00:00.186 SCR: nv=24, tv=3, vns=126, vmem=2467, smem=8192, gmem=488, pmem=1521, tmem=11273

mi-hol, 2025-02-16:
> würde einzelne Variablen mit p: definieren und arrays im Filesystem speichern.

ok, probiere ich.

den code für #load würde ich aber wie folgt verbessern

```
fh=fo("/wd_log.txt" r)
if fh>=0 {
    fra(axhum fr)
    fra(tmpa fr)
    fra(aco2 fr)
    fra(atvc fr)
    fc(fh)
} else {
    print file does not exist - fh: %fh%
    print initialize data expected to be load from file!
}
```
Passt das?

gemu2015, 2025-02-16:
Ja, aber da kommt immer -1 als Fehler raus

mi-hol, 2025-02-16:
was macht der Befehl "dp2"?
Im "Beispiel für das speichern von 4 Arrays" wird er verwendet, aber in  der [scripting Doku](https://tasmota.github.io/docs/Scripting-Language/#e-paper-29-display-with-sgp30-and-bme280) nicht erklärt

gemu2015, 2025-02-17:
doch wird er:
dpx sets decimal precision to x (0-9)
dpx.y sets preceding digits to x and decimal precision to y (0-9), the delimiter also sets the decimal point
character to . or ,

mi-hol, 2025-02-17:
Frage: beim folgenden Text verstehe ich die Bedeutung von "fr" nicht, da "with" im Englischen eine Vielzahl an Bedeutungen hat
Soll es bedeuten, dass es der "file handle" vom fo ist?
Dann wäre statt "with" das Verb "of" korrekt

_fra(array fr) reads array from open file with fr (assumes tab delimited entries)
fwa(array fr (a)) writes array to open file with fr (writes tab delimited entries and line feed at end) the optional a parameter ommits the linefeed for appending arrays fx("fname") check if file fname exists_

nimms mir nicht krumm, aber die syntax der scripting language ist, aus meiner Sicht, nicht konsistent und damit fehleranfällig!

dp(2) statt dp2 wäre konsistent, aber dp(2) dreht das Verhalten laut Text unten, ja total um oder verstehe ich das was falsch?
_dp(x y) sets preceding digits to x and decimal precision to y, the delimter if space sets decimal point character to ., a comma sets the point o _comma__

gemu2015, 2025-02-17:
fr in dem Beispiel ist eine Variable und meint "file refererence"
dp() ist genau dasselbe wie dp mit denselben Parametern nur als "Funktion"

mi-hol, 2025-02-17:
also aus dme open kopierten Text des Handbuchs, erschließt es sich mMn nach nicht, dass das überhaupt eine zulässige Syntax ist :(

dp(6.2) hat übrigens, diese interessante Nebenwirkung

[image]

mi-hol, 2025-02-17:
ich schaffe es leider nicht die Array Variablen in eine Datei zu schreiben
Die Datei bleibt leer!
Deshalb nochmals die Frage: 
In welchem Repository poste ich die Reproduktion (denn mit Otellos Skripten hat es wirklich nichts zu tun!)?

gemu2015, 2025-02-17:
was meinst du denn damit ?

paste dein script hier hinein, markiere es und klicke auf  Code Block "<>"

mi-hol, 2025-02-17:
> was meinst du denn damit ?

ein Zitat macht den Kontext klar, mit so einer offenen Frage kann ich leider nicht die gewünschte Antwort liefern

Mit fehlt die Info in welchem Repository die scripting language entwickelt wird, denn nur dort ist mMn der richtige Ort um über Fehler zu diskutieren. 
Oft kann man sich bei Problemen dann auch schon selbst helfen, wenn man den source code anschaut, statt auf die Hilfe anderer zu warten.

> paste dein script hier hinein, markiere es und klicke auf Code Block "<>"

ich bin Maintainer eines OSS projects auf GitHib, wie es geht weiss ich :)

mi-hol, 2025-02-17:
Mein Problem liegt wahrscheinlich an:

_Subroutine invocations may be nested (each level uses about 600 bytes stack space, so nesting level should not exceed 4). _

die Ausgabe von  "print subroutine: *** iDf  start **" fehlt im log und es gibt keine Hinweise auf eine "call stack exception"!
Ist das möglicherweise eine Auswirkung des C3/C6 Reboot Workarounds?

console log:

```
13:17:07.457 subroutine: *** ldDf  start **
13:17:07.460 SCR: file open failed
13:17:07.461 file: /smldata.txt does not exist - fh: -1.00
13:17:07.462 initialize data expected to be loaded from file!
13:17:07.463 call subroutine: iDf
13:17:07.465 subroutine: *** ldDf  end **
```

code:

```
; -- SUBS --
#ldDf
print subroutine: *** ldDf  start **
;sets decimal precision to 2 digits
; ToDo: report bug!
;dp(6.2) ; issue "00-123" displayed on webpage
;dp(2) ; => no webpage
dp2
fh=fo(fn r)
if fh>=0 {
    print loading arrays from file: %fn%
    fra(sday fh)
    fra(dcon fh)
    fra(dprod fh)
    fc(fh)
} else {
    print file: %fn% does not exist - fh: %fh%
    print initialize data expected to be loaded from file!
    print call subroutine: iDf
    =#iDf
}
print subroutine: *** ldDf  end **
; #ldDf end

#iDf 
print subroutine: *** iDf  start **
if TEST==1 and initNTP==1 {
    ; no active Smart Meter required and 0 values allowed!!"
    ; but some values must not be 0  to have visualization on webpage!
    dcon[doy]=11.1
    dprod[doy]=1.2
    EnFrGrid=2.345
    EnToGrid=3.456
} else {
    print !!! initialize data failed !!!
}
print subroutine: *** iDf  end **
; #iDf end

```

mi-hol, 2025-02-17:
aktuellster zusätzlicher Fehler:

`15:05:42.550 SCR: init error -1`
https://tasmota.github.io/docs/Scripting-Language/#script-init-error-codes nennt aber nur Fehler von -4 bis -7

gemu2015, 2025-02-18:
1.  die Einschränkung bei subroutines ist nur bei esp8266 besonders wichtig, bei esp32 sollte das kein Problem sein.
2.  die voreilenden Nullen machen keinen Sinn und sind auch nicht gedacht für negative Werte, eigentlich nur bei Datum und Uhrzeit relevant. die mit dp gesetzen Werte sind global und gelten fortan für alle numerischen Ausgaben.  Sie lassen sich aber jederzeit lokal überschreiben mit z.B. %0.2var%
3. der Fehler -1 deutet auf einen internen Strukturfehler hin der auf ein syntaktisches Problem in der >D Sektion hindeutet. Wenn du mir die >D Sektion postest kann ich den Fehler finden und den Source Code reparieren.

mi-hol, 2025-02-18:
Danke für die Antworten

> 3\. der Fehler -1 deutet auf einen internen Strukturfehler hin der auf ein syntaktisches Problem in der >D Sektion hindeute

ich habe ihn schon gefunden, dieser und die noch fehlenden Erklärungen für -2 & -3 sollten aber dringend in die offizielle Doku

Offen ist noch:

> ich schaffe es leider nicht die Array Variablen in eine Datei zu schreiben Die Datei bleibt leer! Deshalb nochmals die Frage: In welchem Repository poste ich die Reproduktion (denn mit Otellos Skripten hat es wirklich nichts zu tun!)?

da bisher keine Antwort kam, hier die Reproduktion

[ArrayToFileIssue.zip](https://github.com/user-attachments/files/18852059/ArrayToFileIssue.zip)

gemu2015, 2025-02-19:
1. in der >D section dürfen nur Variablen definiert werden aber nicht als Parameter verwendet werden.
	M:dcon=0 dSave funktioniert nicht
	
2. im prinzip ist dein array save in Ordnung. möglicherweise ist in deinem binary das nötige compile flag nicht gesetzt (extended file support, #define USE_SCRIPT_FATFS_EXT) 

3. für and, or Verknüfungen im if statement gilt das sie entweder auf getrennten Zeilen stehen müssen, oder verklammert sein müssen.
also:

if TEST==1
and initNTP==1 {

oder 
if ((TEST==1) and (initNTP==1)) {

Bin mit sehr wohl bewusst dass in der Dokumentation noch vieles nicht oder nicht ausreichend erklärt ist. Habe gesehen dass du da schon einiges verbessert hast, danke dafür.

mi-hol, 2025-02-19:
Danke Gerhard, ich habe die Änderungen gemacht und teste gerade.

Im scripter (oder in einem separaten "linter") sind mMn dringend Fehlerprüfungen für die obigen Fälle einzubauen, sonst wird jedes script nur mit deinem persönlichen Support funktionieren können.

mi-hol, 2025-02-19:
@gemu2015 Frage: wie erzeuge ich mit rnd eine Zufallszahl mit Wert von -800..+5000  ?
_rnd(x) = return a random number between 0 and x, (seed may be set by rnd(-x))_   der Teil "seed" ist mir unklar.
Ist die Syntax eigentlich an irgendeine andere bekannte Programmiersprache angelehnt?

gemu2015, 2025-02-20:
so erzeugst du deine Zufallszahl:
res=rnd(5800)-800

> Seed: Eine Zahl, mit der eine eindeutige Zahlenfolge generiert werden kann. Somit kann mit demselben Seed dieselbe Zahlenfolge generiert werden. Wird kein Seed übergeben, verwendet die Funktion die aktuelle Systemzeit (Sekundenanzahl seit Mitternacht).

die Scripte sollten für den esp8266 funktionieren mit nur sehr wenig RAM und geringem Verbrauch an Flash. deshalb war das Motto: Befehle so kurz wie möglich halten und so gut wie kein Syntaxcheck. Deshalb kommt es auf die Dokumentation an, damit man Syntaxfehler vermeiden kann.

mi-hol, 2025-03-03:
@gemu2015 ich würde mich über eine Rückmeldung zu den Feature request oben sehr freuen, egal ob gut oder schlecht :)

mi-hol, 2025-03-09:
> in meinen Dropbox download sind auch mehrere Beispiele. (Link in den Docs)

@gemu2015 was meinst Du mit "Link in den Docs" ?
Auf https://tasmota.github.io/docs/Scripting-Language/#image-gallery-of-various-tasmota-scripts steht nur:
"these are some examples of more complex scripts to show what is possible.
complex scripts should be edited with the external source editor as they contain lots of comments and indents.
i will provide the sources later."

gemu2015, 2025-03-10:
ausschnitt aus den Docs mit dem Link, dort sind auch die Scripte

Optional external editor
you may use a special external editor with syntax highlighting to edit the scripts. (mac and pc) you may use any number of comments and indents to make it better readable. then with cmd r the script is transferred to the ESP and immediately started. (all comments and indents are removed before transferring) see further info and download [here](https://www.dropbox.com/sh/0us18ohui4c3k82/AACcVmpZ4AfpdrWE_MPFGmbma?dl=0)

mi-hol, 2025-03-10:
Danke, damit habe ich die Skripte gefunden. Aber sie sind schon extrem gut versteckt.

Was meinst Du zu folgendem Vorschlag:

- neues Repository für die Skripte anlegen (zB als Name gemu2015/Tasmota-Skripte )
- dort die Skripte uploaden
- ich ändere die offizielle Tasmota Dokumentation und ergänze pro Beispiel den link zum source code

gemu2015, 2025-03-11:
Hast du dir mal die Dokumentation zum Smart Meter angesehen ?
Dort sind die Scripte "eingeklappt" und stören beim Scrollen nicht.
Das wäre meiner Meinung nach die bessere Option auf bei der Scripting Doc für alle Scripte.
Da würden auch die sehr grossen Scripte nicht stören.
Es gibt noch ein sehr altes Scripting cookbook, das könnte man alternativ auf den neuesten Stand bringen und dorthin alle Beispiele verlegen.
Habe gerade extrem wenig Zeit und kann mich nicht selbst darum kümmern.

mi-hol, 2025-03-11:
> Hast du dir mal die Dokumentation zum Smart Meter angesehen ?

kann ich so leider nicht beantworten, da ich nicht sicher bin welche der vielen Dokumentation Du meinst. Ich versuch raten immer dadurch zu vermeiden, dass ich eine link zu der von mir gemeinten Referenz poste.

gemu2015, 2025-03-11:
https://tasmota.github.io/docs/Smart-Meter-Interface/

gemu2015, 2025-03-12:
das Fenster kann man beliebig scrollen, da bleibt nichts versteckt.
habe trotzdem mal alle scripting relevanten Sachen in meinen Github account kopiert.
https://github.com/gemu2015/Sonoff-Tasmota/tree/universal/tasmota/scripting

mi-hol, 2025-03-12:
> habe trotzdem mal alle scripting relevanten Sachen in meinen Github account kopiert.

Danke Gerhard, das finde ich eine grosse Verbesserung!

Mit einer Sache hadere ich allerdings noch.
Es ist die mMn unpassende extension .txt für ein Tasmota Skript, das einen ganz speziellen Aufbau und Inhalt haben muss.
Diese Anforderung steht im Gegensatz zu unstrukturiertem Text, der in eine .txt Datei gehört.

Wäre es nicht besser hier die extension .tas zu verwenden?

Falls Du zustimmt, kümmere ich mich gerne um das Umbennen und einen kleinen Patch im ScriptEditor.

gemu2015, 2025-03-12:
hab die Endungen geändert auch im Scripteditor.
Allerdings benutzen 90 Prozent der User keinen Scripteditor und kein VSC plugin. Deshalb sollte bei den einfachen Scripten die Endung .txt bleiben da das dann problemlos mit fast jedem Editor bearbeitet werden kann.

mi-hol, 2025-03-12:
> Deshalb sollte bei den einfachen Scripten die Endung .txt bleiben 

Das würde ich nicht empfehlen, da wir sonst den Vorteil der klaren Identifikation verlieren

> da das dann problemlos mit fast jedem Editor bearbeitet werden kann.

 .tas kann man auch einfach mit Windows Notepad oder jedem anderen Editor verbinden
Im Windows Explorer rechts klick und "öffnen mit" auswählen
In meinem Englischen Windows sieh das so aus

[image]

mi-hol, 2025-04-27:
Fortsetzung in https://github.com/gemu2015/Sonoff-Tasmota/issues/67

closed 2025-04-27 by mi-hol