ottelo9/tasmota-sml-script #7: Jahresverbrauch

Sinned42, 2025-03-11:
Ich habe mir mal Gedanken gemacht über den Titel Jahresverbrauch. Stand jetzt ist es ja der Verbrauch vom 1.1. bis jetzt. Als Verbesserung hätte ich jetzt folgenden Code:

; Tagesverbrauch & Einspeisung Berechnung um Mitternacht
	if ((chg[hour]>0) and (hour==0)) {
		if (day>1) {
			da=day
		} else {
		    ; Monatswechsel
			for tmp (da+1) 31 1
				dpro[tmp]=0
				dcon[tmp]=0
			next
			; monthly values
			mval=EnFrGrid
			mval2=EnToGrid

    yval=0
    yval2=0
			for tmp 1 12 1
				yval+=mcon[tmp]
                yval2+=mcon[tmp+12]
			next
		}
;		if (chg[yr]>0) {
;                Jahreswechsel
;                yval=EnFrGrid
;                yval2=EnToGrid
;		}

Fängt an bei yval=0

Am Anfang eines Monats wird das komplette Array von 1-12 aufsummiert, und man hat dann tatsächlich den Verbrauch für 1 komplettes Jahr. Das ist das, was man i. d. R. braucht, wenn man den Stromanbieter wechselt.

Entsprechend muss es dann noch kleinere Änderungen im Code geben:

; Verbrauch
Tagesverbrauch{m}%2(EnFrGrid-dval)% kWh
Monatsverbrauch{m}%2(EnFrGrid-mval)% kWh
Jahresverbrauch{m}%2(yval)% kWh

; Einspeisung (PV)
Tageseinspeisung{m}%2(EnToGrid-dval2)% kWh
Monatseinspeisung{m}%2(EnToGrid-mval2)% kWh
Jahreseinspeisung{m}%2(yval2)% kWh

Was hältst du davon?

Gruß, Dennis
P. S.: das von mir modifizierte pow2-Script hast du so so umgestellt, dass man nicht mehr energy total resetten kann, ohne dass die Berechnung aus dem Tritt kommt. Mein Script basierte nur auf energy today und energy yesterday. Finde ich persönlich schöner, auch wenn es natürlich ein größerer Eingriff in das Script war. Ich habe es mir wieder umgeschrieben, auch mit der o. g. yval - Geschichte. Bei Interesse lade ich es gerne hoch.

ottelo9, 2025-03-12:
Danke für deine Tipps. Finde ich gut bzgl. des Jahreszählers. Werde ich gerne bei Zeiten implementieren!

Bzgl. dem pow2 Script. Schicke mir gerne dein umgebautes Script. Ich ändere das dann hier um. Einfach als txt Datei hier anhängen.

ottelo9, 2025-03-12:
Ich habe deine Änderung doch nochmal überdacht und bin nun anderer Meinung. Man muss beachten, dass manche den Jahresverbrauch genau so haben möchten, wie es bislang implementiert ist:  "Aktueller Verbrauch in diesem Jahr (bis jetzt)".

In deinem Fall würde man einen prognostizierten Jahresverbrauch haben. Der setzt sich dann aus den tatsächlichen Monatswerten des aktuellen Jahres + aus den Monatswerten des vorherigen Jahres zusammen. Das ist so, da die Werte im Diagramm (Variable mcon[1-12=Verbrauch / 13-24=Einspeisung]) nicht beim Jahreswechsel auf 0 gesetzt werden.

mi-hol, 2025-03-24:
@Sinned42 schön dass Du Verbesserung am Skript machst. 
Aber warum nutzt Du nicht die von git vorgesehenen Prozess "Pull Request (PR)" um die Änderungen sicher und einfach zu sehen?

Ich traue zB keinem code der auf einem file hoster liegt und werde ihn dort nicht ansehen!