ottelo9/tasmota-sml-script #12: [Enhancement] add "include <file_name>" to scripting language, to simplify user script changes

mi-hol, 2025-08-19:
@gemu2015 could the scripting language be enhanced to allow for inclusion of external files?

Current situation:

- user scripts using SML always require adaptions to the specific power meter, but then stay fixed forever or at least a few years
- improvements to user scripts (i.e. [2_SML_Script_Chart_PV_ShellyEmu.tas](https://github.com/ottelo9/tasmota-sml-script/blob/main/2_SML_Script_Chart_PV_ShellyEmu.tas) happen frequently, but currently always require manual changes

Proposed change:

- add a new command "include <file_name>" to scripting language

Benefit:
 
- avoid the need for manual changes before deployment
- reduce skills required to deploy user script changes (i.e. by seniors with very limited computer skills)
- delay for deploment of changes to user scripts can be greatly reduced

Would that enhancement be feasible?

ottelo9, 2025-11-04:
> dazu muss jemand leider alle Zähler in ottelos repo udpaten wie schon beschrieben.

Kein Problem, das werde ich diese Woche machen

gemu2015, 2025-11-04:
@ottelo9 
und wir bräuchten deine binaries mit den neuesten Änderungen drin. Am besten schickst du mir die Quellen dann zu damit wir auf dem gleichen Stand sind.  Die JavaScript Erweiterung sollte aber unbedingt mit #ifdef abgesichert sein.

gemu2015, 2025-11-04:
@mi-hol 
hier ein C3 binary allerdings ohne die JavaScript Änderung.

[tasmota32c3-4M.bin.zip](https://github.com/user-attachments/files/23329823/tasmota32c3-4M.bin.zip)

gemu2015, 2025-11-04:
habe jetzt ein macro gebaut mit dem man die Javascript Meter abholen kann. Das funktioniert auf allen CPUs
bin jetzt am überlegen was dann passieren soll.
ich könnte einfach das script überschreiben so wie bei ottelos Lösung.
Jedes script müsste dann aber den download Befehl enthalten.
```
>W
smlpd("https://raw.githubusercontent.com/ottelo9/tasmota-sml-script/refs/heads/main/smartmeter" "select a meter")
```

ottelo9, 2025-11-04:
Dann würde ich die Kombi mit meinem "JavaScript DropDown Menü" zur Auswahl des Hauptscripts  (Edit Script Seite) und deinem neuen smarten Macro zur Auswahl des Descriptors + Pin wählen.

Die Javascript Geschichte hatte ich bereits bei dir im scripter via #defines drin, muss mir das aber nochmal anschauen, weil ich OTTELO in die #defines geschrieben hatte ...

ottelo9, 2025-11-04:
@gemu2015 
Ich habe mir nun ein angepasstes Image gebaut. Ich kann testweise ein "Hauptscript" laden z.B. http_select_script.tas und dann auf der Hauptseite lade ich ein smartmeter z.B. MT175 und wähle die korrekten Pins. Aber ich bekomme keine Daten angezeigt.

[image]

[image]

Ich habe auf meinem Repo einen neuen Testordner für Testscripte erstellt und die Pfade bei mir angepasst:
https://github.com/ottelo9/tasmota-sml-script/tree/main/smartmeter_test

Files:
[tasmota32c3_ottelo.zip](https://github.com/user-attachments/files/23344518/tasmota32c3_ottelo.zip)
[user_config_override.h.txt](https://github.com/user-attachments/files/23344531/user_config_override.h.txt)
[scriptlist.h.txt](https://github.com/user-attachments/files/23344536/scriptlist.h.txt)
[xdrv_10_scripter.ino.txt](https://github.com/user-attachments/files/23344538/xdrv_10_scripter.ino.txt) <- deine neuste Scripter inkl. meinem Dropdown

die xsns_53_sml.ino von dir habe ich auch

gemu2015, 2025-11-04:
Ja da fehlt noch was.
In >B müssen die Pins gesetzt werden vor dem sensor53 r
rxpin=rxind-1
txpin=txind-1

Und in >S bei chg noch jeweils sensor53 r

ottelo9, 2025-11-06:
@gemu2015 Habe es auf meinem ESP32-C3 Lesekopf ausprobiert, funktioniert gut. Ich habe noch ein paar Namen geändert im Scripter inkl. kleiner Beschreibung, kannst du das noch übernehmen? Eine Sache gibt es noch, die die User evtl. verwirren könnte. Das DropDown im Script zeigt nach einem Seitenreload wieder den ersten smartmeter aus der Liste an, nicht den gewählten. Evtl. kannst du das noch ändern, falls du Zeit hast?

[xdrv_10_scripter.ino.txt](https://github.com/user-attachments/files/23383454/xdrv_10_scripter.ino.txt)

gemu2015, 2025-11-06:
@ottelo9 
ok, hab den selector addiert damit er sich den Eintrag merkt.
Anmerkung: bezüglich der Javascript selection im edit script weist du schon dass das mit grossen Skripten auf dem ESP8266 nicht funktioniert wegen zu wenig RAM.
@mi-hol 
seit kurzem ist die scripting doku gecrasht. finde keine Ursache in den letzten commits. Kannst du mal schauen ob du was findest ?

gemu2015, 2025-11-06:
merkwürdig, das hatte heute morgen geholfen.
du brauchst keinen reboot sondern einen refresh der Seite.
weis gerade nicht wie das zu stande kommt.
du siehst das Problem da der Sekunden Zähler stehen bleibt.

gemu2015, 2025-11-06:
ja bei mir war halt pin 17 drin.  da ich bisher keine Deskriptoren mit %0rxpin% hatte war das nicht abgestürzt
ist schwierig den Fehler zu finden warum es nicht upgedated wird.  Der SML Treiber wird aber immer richtig initialisiert, nur die WEBUI wird nicht aufgerufen.

gemu2015, 2025-11-06:
Ja klar konnte man schon immer einen falschen pin angeben, aber jetzt mit dem pulldown ist es leichter möglich 
Soweit ich weis gibt es innerhalb von tasmota eine Liste welche CPU welche Pins hat. Mal sehen ob ich das verwenden kann

mi-hol, 2025-11-07:
> [@mi-hol](https://github.com/mi-hol)
> seit kurzem ist die scripting doku gecrasht. finde keine Ursache in den letzten commits. Kannst du mal schauen ob du was findest ?

das behandeln wir besser in https://github.com/gemu2015/Sonoff-Tasmota/issues/67#issuecomment-3501932398

mi-hol, 2025-11-07:
Ich finde die Reihenfolge der Dialogboxen für Usability wie folgt am besten

[image]

Ist das die letzte Version?

gemu2015, 2025-11-07:
ja

gemu2015, 2025-11-07:
habe getestet mit esp8266 und esp32-c3 ohne jede Probleme mit Safari und Chrome auf macos

mi-hol, 2025-11-10:
> D.h. am Anfang Daten aus dem Filesystem laden und in die "nicht permanenten" Arrays schieben

@ottelo9 genau das hatte ich auch mal angefangen, aber leider bisher nicht fertig stellen können.
Falls Du meinen Versuch anschauen magst, kann ich ihn gerne hier anhängen oder Dir Zugriff auf ein privates Repo geben.

gemu2015, 2025-11-10:
nur zur Info: hab den Fehler in der scripter doku gefunden. 
alles wieder OK

mi-hol, 2025-11-11:
> hab den Fehler in der scripter doku gefunden.

war das fehlende "escape" für <> ?

ottelo9, 2025-11-24:
Probiere das aus dem [Release](https://github.com/ottelo9/tasmota-sml-images/releases/tag/V15.0.1_251026).

closed 2026-03-20 by ottelo9

mi-hol, 2026-05-10:
> einen #include gibt es tatsächlich nicht, ist auch bisher nicht geplant (Anmerkung: für TinyC).

@gemu2015 Danke für die Implementierung in  
"TC_RELEASE 1.5.2 — #include "file.tc" directive (IDE-side preprocessor).
The IDE compiler now recognizes #include "file.tc" and inlines the content before compile." 

@ottelo9 würden sich die 2 Skripte in https://github.com/ottelo9/tasmota-sml-script/tree/main/tinyc nicht zum Testen des neuen Features anbieten?